labapart/gattlib #290: can I scan duplicates?

jingoro2112, 2025-03-06:
gattlib is great, and does everything I need with one exception: I would like to scan and receive **ALL** beacon reuquests even if they have been seen before.

This is so I can read data from the name/man data without having to connect.

I know gattlib is a wrapper for bluez but I don't know enough about that interface to figure out where/what I need to do to get this functionality. No matter what I try I only ever get devices enumerated a single time.

RavenNil, 2025-04-09:
i am facing the same scene, i have tried the below codes:

```c
while(1){
  gattlib_adapter_open

 // TIMEOUT 4 seconds
  gattlib_adapter_scan_enable_with_filter    
  gattlib_adapter_scan_disable

  gattlib_adapter_close
}
```
At the beginning, the callback is triggered continuously, but after a while, the callback stops being triggered. However, the infinite loop above is still executing.

RavenNil, 2025-04-10:
i think i found the code to edit.
at `on_interface_proxy_properties_changed` in `gattlib_adapter.c`

edit 

```c
enum _gattlib_device_state old_device_state = gattlib_device_get_state(gattlib_adapter, proxy_object_path);

if (old_device_state == NOT_FOUND) {
    if (has_rssi || has_manufacturer_data) {
        int ret = gattlib_device_set_state(gattlib_adapter, proxy_object_path, DISCONNECTED);
        if (ret == GATTLIB_SUCCESS) {
            gattlib_on_discovered_device(gattlib_adapter, device1);
        }
    }
 }
```
TO

```c
// enum _gattlib_device_state old_device_state = gattlib_device_get_state(gattlib_adapter, proxy_object_path);

// if (old_device_state == NOT_FOUND) {
    if (has_rssi || has_manufacturer_data) {
        int ret = gattlib_device_set_state(gattlib_adapter, proxy_object_path, DISCONNECTED);
        if (ret == GATTLIB_SUCCESS) {
            gattlib_on_discovered_device(gattlib_adapter, device1);
        }
    }
//  }
```
AND start the scan indefinitely.